bigtreetech/BIGTREETECH-SKR-mini-E3 #720: Example config for CR-10 with BTT SKR Mini E3 V3

Velo555, 2023-02-04:
Will there be a example config for CR-10 with V3 board installed. It is sold stating that it is for E3/E5/CR-10. But it only has examples for E3 and E5. The latest example for CR-10 is V2.

I would like an example config to make it less intimidating to new users.

**Additional context**
Right now I am using paid firmware from TH3D. I do love this board it makes my CR-10 so quite.

itstheburns, 2023-02-07:
I second this. I just purchased this board with the tft35 panel and unable to customize for my CR-10.

radek8, 2023-02-09:
https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Creality

Velo555, 2023-02-10:
That link is not helpful it leads me to the example I was talking about. They only have an example config for the v2 board not the v3 board.

radek8, 2023-02-10:
To compare two files, we use Notepad++ with the Compare plugin
https://notepad-plus-plus.org/

Velo555, 2023-02-10:
Now that's what I'm talking about!! Of course because of my luck I am about 95% done.

Velo555, 2023-02-14:
> I second this. I just purchased this board with the tft35 panel and unable to customize for my CR-10.

Itstheburns - I just finished compiling one for it. Do you want it?

Velo555, 2023-02-14:
Which tft35?

SrgWhoopsie, 2023-02-23:
yall can compare in vscode aswell, and also instantly switch the difference in file with one click FYI

cappelletto, 2023-05-10:
@Velo555 did you manage to compile it for the CR-10?
If you feel so, could you share it or at least point the major issues/changes when compared to V2 (assuming you used it as reference).
Still undecided between E3 V2 and V3, so I wanted to have a better picture about the support for CR-10 (pretty old one, stock v1)

CHeers

Velo555, 2023-05-11:
I did compile it. I compared the CR-10 firmware for skr mini e3 v2 versus CR-10 S5 with skr mini e3 v3. I will have to see if I kept my notes. Right now I am running firmware from Nic Wilson (Facebook group). His does faster bed probing. I need to sit down and look at what the differences are.

cappelletto, 2023-05-11:
Thanks for the info, I will use those two firmware configurations as references. If you happen to find your notes, even better.
I wasn't aware of the N. Wilson (fb group) firmware, I'll take a look at it. It sounds like a good-to-have backup plan,
Cheers

cappelletto, 2023-05-11:
@Velo555 follow-up question: was it using the stock panel or a new tft one?

Velo555, 2023-05-11:
Stock

cappelletto, 2023-05-14:
Thanks, I finally managed to do it using both the reference configurations you mentioned and extra info from [https://3dwork.io/en/skr-mini-e3-v3/](https://3dwork.io/en/skr-mini-e3-v3/).

Enabled the bed leveling (bilinear) and couple of extra entries to the menu. So far so good, and the best thing: it is so silent! Def. worth the upgrade.

Velo555, 2023-08-22:
Once stable you can make some of the changes in this video. Use caution. 
https://youtu.be/PFuz8915GCs

cappelletto, 2023-08-22:
Ok, I was considering using a non-contact probe but not sure yet about which route take - the video you shared has some interesting numbers.
3 days ago I switched from an old faithful glass bed top to a double sided magnetic bed sheet and results are more robust against potential miscalibrations so I will postpone that mod for a couple of months.