bigtreetech/BIGTREETECH-SKR-mini-E3 #721: [ISSUE] SKR Mini E3 V2.0 Bad Incorrect TH0 Thermistor value

Sporkforce, 2023-02-07:
Finally got my SKR Mini E3 V2 installed after a few months of saving up for a BTT TFT35 (wanted to do the upgrade all at once) so the official BigTreeTech Store I purchased this from won't help me because of time elapsed. Everything works fine except the Hot End Thermistor value. At ambient it is reliably reading 20°C above actual temp. Problem persists through 3 separate thermistors. Firmware config is correct and thermistor designation is flawless on my other board using the exact same thermistors. Voltage at the pins are supposed to be 3.3v, they are 3.0. This seems like a known issue, but I'm not sure how to fix this. From the schematic it seems that part CG0603MLC-3.3LE is the varistor dictating circuit voltage for TH0. One complaint seems to show this specific varistor detonating as well (Otherwise I'd probably just make a custom resistance table and call it a day). Is it even safe to use these boards? Anyone have experience with this issue? Am I going to have to try a DIY SMD repair because I didn't catch the manufacturer's own error quickly enough?
[image]
[image]

SIXM6N, 2023-04-17:
Wow this is exactly the same issue I have. At room temp the hot end reads 47c. The bed sensor is reading spot on.

Fallstern, 2024-07-07:
@Sporkforce Did you fix this problem somehow? Have the same problem