bigtreetech/BIGTREETECH-SKR-mini-E3 #722: Ender 3v2 Skr mini e3v3 wont print

Niklas0021, 2023-02-08:
Hello,
after a long try and error i have got my Ender 3v2 to work with the skr mini e3 v3 but the printer behaves strange. I wanted to print the calibration cube. When i start the print, the printer homes x y z (i have disabled the CR Touch because i could not get it to work so i went back to the original z endstop). Then the printer waits for maybe 3 seconds, moves a little bit (like it would print really small and really fast for about 2 seconds), the extruder gear goes a bit forwards and backwards and then ends the print saying print done in 00:00 minutes and goes to its end position and starts to cool down. Anyone had this problem before and maybe knows the problem ?
would be thankful for anyone helping me.
Greetings Niklas

nocturnalit, 2023-02-14:
I’m having the exact same issue, I had multiple boards to test and thinking it was a board issue I swapped them and got the same result. I’ve tried several firmwares including the one from the bigtreetech GitHub page and they all do the same thing. Homes and heats up ok but once it reaches temperature then nothing, progress bar runs for a few seconds and thinks the job is done and starts cooldown. 

Everything but printing works and it’s got me really frustrated to the point I will be going back to my v2 board.

aldoguzman97, 2023-02-25:
I encountered the same issue after an update switching from software to hardware serial ports. This resolved the significant movement lag that was previously present in the xyz axis, but caused other issues.