bigtreetech/BIGTREETECH-SKR-mini-E3 #723: No printer attached

Korpz58, 2023-02-11:
Hi all! Very new to this so please bear with me as I have ZERO idea what's going on. I bought a ender 3 and replaced all the things with a skr mini e3 v3, tft35 e3 v3.0, followed all the instructions to a tee, and now it says "no printer attached". I have attempted to "flash" firmware, but it does not seem to be doing anything. I have yet to be able to use the printer at all, only made 1 straight line with it (old board crealty v4.2.2), and then it would freeze, that's why I changed the board out from what everyone was saying. I again have no idea what I'm doing with this, but I'm ready to throw it through the window to say the least. If anyone could please help me out here, I would be eternally grateful.
Thank you

smumriak, 2023-02-11:
change the partition size of your SD card to exactly 1024 MB. anything higher didn't work for me. use standard FAT32 filesystem (I didn't try fat16, but some people claim it works too)

radek8, 2023-02-16:
You must compile Marlin with serial port 2 enabled
for example:
#define SERIAL_PORT 2

the speed must be the same in Marlin and TFT
for example 115200
#define BAUDRATE 115200

On TFT in config.ini
serial_port:P1:6 P2:0 P3:0 P4:0

bens888, 2023-04-12:
I have the issue with "No Printer Attached".  Current hardware is TFT35-E3 V3.0.1 with motherboard SKR-Pro v1.2.
 
Marlin firmware 2.1.1 in configuration.h file :

#define SERIAL_PORT 6 //1 //-1 //1 //0 // previous tried ports

#define BAUDRATE 115200
#define SERIAL_PORT_2 -1//1 //-1 //-1

also when in Marlin mode, the main screen is showing "TMC connection error".

Not sure what to try next...

radek8, 2023-04-12:
The serial port number depends on the Marlin motherboard used and the pins used....
This is not a lottery :-)

radek8, 2023-04-12:
The TMC connection error is displayed if you power the board from USB, or if you do not have the TMC drivers configured on the UART mod

bens888, 2023-04-12:
Have not powered the motherboard with USB.
UART connection:
[image]

bens888, 2023-04-12:
for all five TMC2209 drivers.

radek8, 2023-04-13:
Send a detajl photo to TMC from both sides

bens888, 2023-04-14:
any comments on the TMC2209 driver?

radek8, 2023-04-14:
My TMC2209 drivers have a jumper to communicate via UART. If they didn't communicate over the serial port to someone, there was a problem with the jumper.
Unfortunately, your TMCs have a different cast. That's why I can't say where they have the switch to UART mode.
I hope they are okay.

What is your response to the command
M122

bens888, 2023-04-15:
No Printer Attached, so cannot issue any commands from the TFT, have not tried with the USB interface on the SKR Pro yet.

mdcasa, 2023-05-09:
If the firmware-bltouch.bin does not flash, is the only alternative to compile it myself? I do see the post about partition size which I will try also. The file "firmware.bin" works fine, but the "bltouch" enabled one will not work at all.