bigtreetech/BIGTREETECH-SKR-mini-E3 #724: BTT TFT35 E3 V3.0.1 "no printer attached" - BTT SKR Mini E3 V3.0

TheEntropy986, 2023-02-17:
Hi,
The TFT35 E3 v3.0.1 connected to SKR3-Mini-E3 V3.0 shows "no printer attached"

I attempted to follow along with [this](https://github.com/bigtreetech/SKR-3/issues/24) and the marlin changes that user radek8 suggested [here](https://github.com/bigtreetech/SKR-3/issues/24#issuecomment-1254006810).
I also attempted to use @radek8 's .zip for the GD_TFT35 firmware that was "adjusted" for the UART3 alternate configuration. Pins are currently configured for the alternate configuration connection. RS232 wiring has been confirmed with a multimeter to NOT be broken. Pinouts on SKR board and TFT35 verified in correct configurations from BTT github hardware images. 

When using SERIAL_PORT_3 as defined in the configuration.h file, it caused the SKR mini e3 to not display anything over "marlin mode". Marlin mode works and controls printer with uncommented. 

https://github.com/TheEntropy986/BTT_TFT35/blob/main/329671257_2114169928973142_4091880034721631840_n.jpg

https://github.com/TheEntropy986/BTT_TFT35/blob/main/330642767_1233412877612889_6877404717543747953_n.jpg

https://github.com/TheEntropy986/BTT_TFT35/blob/main/330916876_1212425312705457_2603581474396460524_n.jpg

https://github.com/TheEntropy986/BTT_TFT35/blob/main/330924849_9679419795417128_8301362167759486807_n.jpg

https://github.com/TheEntropy986/BTT_TFT35/blob/main/331231519_506657734754592_4076458581046289942_n.jpg

https://github.com/TheEntropy986/BTT_TFT35/blob/main/331251192_6230518776961236_242686320731766732_n.jpg

https://github.com/TheEntropy986/BTT_TFT35/blob/main/331304058_2383760815124978_7439339999094046021_n.jpg

Using the marlin bugfix 2.0.x firmware with the following changes:

    file: Configuration.h

```
#define STRING_CONFIG_H_AUTHOR "(ThunderClap)" // Who made the changes.
#define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V3_0
#define SERIAL_PORT -1 // Enable TFT serial port
#define BAUDRATE 115200
#define SERIAL_PORT_2 2 // Enable USB emulated serial port 
#define BAUDRATE 115200
//#define SERIAL_PORT_3 3 - with this uncommented marlin mode non functioning
//#define BAUDRATE 115200 - with this uncommented marlin mode non functioning
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
#define E0_DRIVER_TYPE TMC2209
#define EEPROM_SETTINGS     // Persistent storage with M500 and M501
#define SDSUPPORT
#define CR10_STOCKDISPLAY
```

    file: Configuration_adv.h
```

#define LONG_FILENAME_HOST_SUPPORT   
#define AUTO_REPORT_SD_STATUS         // Auto-report media status with 'M27 S<seconds>'
#define SDCARD_CONNECTION ONBOARD
#define BABYSTEPPING
#define EMERGENCY_PARSER
#define SERIAL_FLOAT_PRECISION 4
#define AUTO_REPORT_POSITION
#define M115_GEOMETRY_REPORT
#define M114_DETAIL         // Use 'M114` for details to check planner calculations
#define REPORT_FAN_CHANGE   // Report the new fan speed when changed by M106 (and others)
#define HOST_ACTION_COMMANDS
#define HOST_PROMPT_SUPPORT           // Initiate host prompts to get user feedback
#define HOST_STATUS_NOTIFICATIONS   // Send some status messages to the host as notifications
```

TFT35 E3

Firmware used: [BIGTREE_GD_TFT35_V3.0_E3.27.x.bin](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_GD_TFT35_V3.0_E3.27.x.bin) with the default config.ini file, The info screen shows GD_TFT35_E3_V3.0

Also attempted (and currently have loaded) [BIGTREE_GD_TFT35_V3.0_E3.27.x.bin](https://github.com/bigtreetech/SKR-3/files/9618967/BIGTREE_GD_TFT35_V3.0_E3.27.x.zip) with @radek8 's adjusted changes for TFT.

TheEntropy986, 2023-02-17:
Current wiring configuration:
TFT35               SKR Mini 

- RST (RS232)     RST (TFT-GPIO)
- RX (UART3)      RX (TFT-GPIO)
- TX (UART3)      TX (TFT-GPIO)
- GND (RS232)   GND (TFT-GPIO)
- +5v (RS232)    +5v (TFT-GPIO)

radek8, 2023-02-17:
Hello
Can you send the Marlin configuration files with the current configuration?
configuration.h
configuration_adv.h
Thank you

radek8, 2023-02-17:
Check the serial cable if it has well pressed connectors and the cable is electrically conductive.
If everything is fine and the display still does not work, try swapping the RX and TX pins on the display on the UART3 connector.

tthenrie, 2023-02-18:
Following this thread. Same issue with SKR Mini E3 V3.0. I believe it is a faulty or nonexistent bootloader. My board will not flash Klipper, but will connect with octopi with the base Marlin firmware that comes installed on the board from the factory. I cannot flash updated Marlin firmware.

TheEntropy986, 2023-02-18:
> Hello
> 
> Can you send the Marlin configuration files with the current configuration?
> 
> configuration.h
> 
> configuration_adv.h
> 
> Thank you

Where would you like me to send them?

TheEntropy986, 2023-02-18:
> Check the serial cable if it has well pressed connectors and the cable is electrically conductive.
> 
> If everything is fine and the display still does not work, try swapping the RX and TX pins on the display on the UART3 connector.
> 
> (with modified FW)

The cable has been tested for continuity with a multimeter. The cable is electrically conductive end-to-end. 

I will try swapping the RX and TX cables.

TheEntropy986, 2023-02-19:
> Following this thread. Same issue with SKR Mini E3 V3.0. I believe it is a faulty or nonexistent bootloader. My board will not flash Klipper, but will connect with octopi with the base Marlin firmware that comes installed on the board from the factory. I cannot flash updated Marlin firmware. From my searches I can see this issue has been around for a loooong time on the SKR Mini boards.

I can flash updated Marlin firmware on the SKR Mini E3 V3.0. The issue I am having is that the TFT35 and the SKR are NOT talking over serial connection. My `#CR10_STOCKDISPLAY` configuration in "Marlin mode" is working from the SKR EXP1 to the TFT EXP3 connections just fine.

ThunderClap86, 2023-02-19:
> (with modified FW)

The TFT firmware or the Board Marlin FW? I used your "modified FW" file on the previously linked issue (you uploaded a .bin and a .ini file [here](https://github.com/bigtreetech/SKR-3/issues/24#issuecomment-1254006810))

ThunderClap86, 2023-02-21:
> Check the serial cable if it has well pressed connectors and the cable is electrically conductive.
> 
> If everything is fine and the display still does not work, try swapping the RX and TX pins on the display on the UART3 connector.
> 
> (with modified FW)

Okay, sorry for the delay, I am attempting this again today. I will post the update. 

In your config.ini for the screen, does the port for UART3 need to be enabled? The one I pulled from the previous ticket has only the main serial port enabled and the others are zero. 

Should the RX of one device go to the TX of the other? (RX screen to TX board, TX screen to RX board?)

ThunderClap86, 2023-02-21:
> Try swapping the RX and TX pins on the display on the UART3 connector. (with modified FW)

THIS WORKED. SOLVED. The RX and TX wires were swapped and the serial connection started right away. The other thing I did was go into the config.ini and enable P2:6 and P3:6 `101 serial_port:P1:6 P2:0 P3:0 P4:0`.

ThunderClap86, 2023-02-21:
For those having the same issue with the SKR mini E3 V3.0 and the TFT35 v3.0.1 (cant say this works for ANY other configuration, follow the guide [here](https://github.com/bigtreetech/SKR-3/issues/24#issuecomment-1254006810) but DO NOT enable the #define SERIAL_PORT_3 3 as suggested in your marlin configuration.h.

SKR Mini E3 V3.0:

    1. Flash a marlin 2.1.x configuration to the board (BTT site says something about it being needed, cant find it again) *configuration.h settings in step 4 below IMPORTANT.

- Marlin settings

        ```
#define SERIAL_PORT 2

        #define BAUDRATE 115200

        #define SERIAL_PORT_2 -1

        #define BAUDRATE_2 115200

        ENSURE #define CR10_STOCKDISPLAY
``` Is the ONLY screen uncommented (enabled)

2. Ensure serial wire is NOT broken (continuity test with multimeter is best)
3. Ensure serial wire between TFT is wired like this (left to right with stepper plugs as "up")
       - RST(loose wire)>RX>TX>GND>5v

    

TFT35 E3 V3.0.1 (GD Controller chip):

1. Before going nuclear and re-wiring try this
2. Flash newest firmware (GD or STM controller needs to be confirmed first) from BTT github.
3. Undo the RX and TX wires from the JST cable (screen side plug) and swap their positions.
5. Power cycle the system
**Nuclear option**

1. Undo the RX and TX wires from the JST cable
2. Place the RX wire from the board into the TX of the UART3 connection (Far left with EXP connections as top)
3. Place the TX wire from the board into the RX of the UART3 connection (middle)
4. Plug JST connection back into the RS232 port at upper right, now missing two wires. 
             --Looks like RST>BLANK>BLANK>GND>5v
6. Plug the JST connector (or bare wire to test, do not short this out!!!!) into the UART3 port.
7. Flash the firmware from [here](https://github.com/bigtreetech/SKR-3/files/9618967/BIGTREE_GD_TFT35_V3.0_E3.27.x.zip) *WARNING - GD controller chips ONLY* to the TFT35
8. Format SD card
9. Place BIGTREEGD_TFT35_V3.0_E3.27.x.bin file on SD
10. Open config.ini file and CHANGE line 101 to be:

            `101 serial_port:P1:6 P2:6 P3:6 P4:0` really "P3:6" should be all you need (UART3 port) but for some reason mine would not work without P2 also placed at "6" (6= 115200 baudrate)
11. Save the changed config.ini file to the SD card along with the step 2 file above.
12. Power cycle the TFT35 and SKR while connected together.

closed 2023-02-21 by ThunderClap86