bigtreetech/BIGTREETECH-SKR-mini-E3 #728: [Display BTT 12864 V2.0 on BTT SKR E3 Mini V3] (How to adapt the BTT 12864 v2 to SKR E3 Mini V3)

PaulSolodovnikov, 2023-03-04:
Recently i bought the BTT SKR E3 mini V3 and a BTT mini 12864 V2.0 display 
[https://github.com/bigtreetech/MINI-12864/tree/master/mini12864_v2.0](BTT mini 12864 V2.0). Not the best decision, but done.

Found similar issue [https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/650](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/650) of BTT SKR E3 mini V3 and a BTT 12864 V1 display.

But pinout on BTT 12864 V1 and V2 are different.

Can anyone help me to make it work? )
[image]

radek8, 2023-03-04:
Yes, bad decision and bad HW combination. :-(
The pinout image defines different pins for a different driver than the V1.0 version.
But the user manual says to use the same driver:
#define BTT_MINI_12864_V1
BTT is quite chaotic when it comes to documentation and manuals...

Use the advice from the post https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/650
Use the same pins in the connector.
Orient yourself by pin GND and 5V
It might work...

radek8, 2023-03-04:
[image]

PaulSolodovnikov, 2023-03-10:
In short I've configured Marlin and made the connections as if I had V3 Mini Screen adapter for Voron [https://github.com/VoronDesign/Voron-Hardware/tree/master/SKR-Mini_Screen_Adaptor](url)

Marlin 2.1.2, Sample comfig files for Creality Ender 3 as it uses BTT SKR mini E3 V3.

configuration.h
//#define CR10_STOCKDISPLAY //comment this

#define FYSETC_MINI_12864_2_1    // Type A/B. NeoPixel RGB Backlight //uncomment this

 #define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V3_0
 #define SKR_MINI_SCREEN_ADAPTER //add this line right after the previous

 #define NEOPIXEL_TYPE          NEO_RGB //change value to NEO_RGB

Hardware connections made according to Marlin/src/pins/stm32g0/pins_BTT_SKR_MINI_E3_V3_0.h SKR_MINI_SCREEN_ADAPTER section and checked with adapter layout https://github.com/VoronDesign/Voron-Hardware/blob/master/SKR-Mini_Screen_Adaptor/SRK%20Mini%20E3%20V3.0/Images/PCB_Layout.png

[image]

Connections made:
BOARD ----- SCREEN
EXP1 GND ----- EXP1 GND
EXP1 5V ----- EXP1 5V
PB5 ---- PB5
PA9 ---- PA9
PA10 ---- PA10
PB9 ---- PB9
PB8 ---- PB8
PD6 ----- PD6
SCK ----- SCK
MOSI ---- MOSI
MISO ----- MISO
PA15 ----- PA15
EXP 1 pin4 RESET ----- EXP 2 pin8 --

Works and starts every time.
As expected no beeper or display SD-card.
Need some time to test...

radek8, 2023-04-29:
Gentlemen, your nanology on the rake is out of place.
  If I buy a small city car and then buy SUV wheels from the same manufacturer. Also, will you be angry at the car manufacturer for making incompatible wheels with your car?
BTT, MKS and others make different components for different printers and it's up to you which components you buy.
We have a saying: Measure twice, but cut once. :-)

nghisus, 2023-05-01:
> @PaulSolodovnikov I followed your instructions and I got the display to work.  Might I add a few things for total noobs such as myself?

Look for the Ender 3 Profile under Configurations-release-2.1.2 > config > example
Make the changes to this Configuration.h file and save it.

Now open Marlin-2.1.2 as a folder.
Delete the Configuration.h and Configuration_adv.h

Drag the modified Configuration.h tab and drop it into the "Marlin" folder on the tree.
Open the Configuration_adv.h file from Configurations-release-2.1.2 for Ender 3, drag and drop it to "Marlin"
From that same folder also do the same for Bootscreen and Statusscreen (also to "Marlin")

Go to platformio.ini and change:
default_envs = STM32G0B1RE_btt

Build.  

So Screen will be stuck on on the firmware version on first bootup.  Turn off, remove SD card, turn on and it should function normally and display Ender 3, build version, then status screen.

Thank you so much for this information.
I will now try to make it work with a CR10S Pro V1.  Would it be ok if I contacted you for help from time to time?

Thanks again!

R2MGaming, 2023-10-03:
I'm starting to have trouble with my Mini 12864 display on my BTT SKR Mini E3 V2 board using an adapter for it. Is there a good Marlin configuration for that?

Tholitz1124, 2023-10-31:
Just got mine working. Everything is good except for the SD card slot. I don't really care much about that since I have this on OctoPrint. Soon to move over to Klipper
[image]

R2MGaming, 2023-10-31:
I already got a touch screen display for mine so I don't need to go through any trouble anymore. Thanks.

veluska, 2023-12-16:
i have anet et4 pro 
i install btt skr mini e3 v3.0.1 with btt mini display12864 v2
please be aware for the pin of the version, the skr v.3.0 is diffrent with v3.0.1 
also btt mini display

i use klipper , i get klipper.bin from the github (go to folder v.3.0.1, NOT folder v3.0)

video :
[https://www.youtube.com/shorts/BrisqOmM5BY](url)

skr mini e3 v3.0.1 pinout :

[image]

for mini 12864 v2.0  pinout,you can go to the github with v2.0 version,

``
the printer.cfg is
[include mainsail.cfg]

[stepper_x]
step_pin: !PC11
dir_pin: PC12
enable_pin: !PC10
microsteps: 16
rotation_distance: 40
endstop_pin: ^!PB5
#endstop_pin: ^!PC0
#endstop_pin: tmc2209_stepper_x:virtual_endstop
#SET_TMC_FIELD STEPPER=stepper_x FIELD=SGTHRS VALUE=255
homing_retract_dist: 0
position_min: -20
position_endstop: 0
position_max: 220
homing_speed: 20

[tmc2209 stepper_x]
uart_pin: PC7
tx_pin: PC6
uart_address: 0
run_current: 0.580
stealthchop_threshold: 999999

[stepper_y]
step_pin: !PB12
dir_pin: !PB10
enable_pin: !PB13
microsteps: 16
rotation_distance: 40
endstop_pin: ^!PB6
#endstop_pin: tmc2209_stepper_y:virtual_endstop
#SET_TMC_FIELD STEPPER=stepper_y FIELD=SGTHRS VALUE=255
homing_retract_dist: 0
position_min: -20
position_endstop: 0
position_max: 220
homing_speed: 20

[tmc2209 stepper_y]
uart_pin: PC7
tx_pin: PC6
uart_address: 2
run_current: 0.580
stealthchop_threshold: 999999

[stepper_z]
step_pin: PB1
dir_pin: !PB0
enable_pin: !PB2
microsteps: 16
rotation_distance: 8
endstop_pin: ^!PB7
homing_retract_dist: 0
position_endstop: 0.0
position_max: 250

[tmc2209 stepper_z]
uart_pin: PC7
tx_pin: PC6
uart_address: 1
run_current: 0.580
stealthchop_threshold: 999999

[extruder]
step_pin: PC2
dir_pin: !PC1
enable_pin: !PC3
microsteps: 16
rotation_distance: 33.500
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PA15
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC5
control: pid
pid_Kp: 20.375
pid_Ki: 0.844
pid_Kd: 123.016
min_temp: 0
max_temp: 250

[tmc2209 extruder]
uart_pin: PC7
tx_pin: PC6
uart_address: 3
run_current: 0.650
stealthchop_threshold: 999999

[heater_bed]
heater_pin: PB3
sensor_type: ATC Semitec 104GT-2
sensor_pin: PC4
control: pid
pid_Kp: 70.721
pid_Ki: 1.981
pid_Kd: 631.118
min_temp: 0
max_temp: 125
max_temp: 130

[heater_fan heatbreak_cooling_fan]
pin: PC8

[heater_fan controller_fan]
pin: PA8

[fan]
pin: PC9

[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32f401xc_h723-if00

[printer]
kinematics: cartesian
max_velocity: 500
max_accel: 3000
max_z_velocity: 12
max_z_accel: 50

[display]
lcd_type: uc1701
cs_pin: PD2
a0_pin: PC0  #a0 is LCD_A0
rst_pin: PB4
encoder_pins: ^PA10,^PA9
click_pin: ^!EXP1_1
contrast: 63
spi_software_sclk_pin: PA5
spi_software_mosi_pin: PA7
spi_software_miso_pin: PA6

[neopixel btt_mini12864]
pin: PB14
chain_count: 3
color_order: RGB
initial_RED: 0.4
initial_GREEN: 0.05
initial_BLUE: 0.0 

[board_pins]
aliases:
# EXP1 header
    EXP1_1=PB15, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PD2, EXP1_9=<GND>,
    EXP1_2=PB14, EXP1_4=<RST>, EXP1_6=PB4,  EXP1_8=PC0, EXP1_10=<5V>

[gcode_macro START_PRINT]
gcode:
    {% set BED_TEMP = params.BED_TEMP|default(60)|float %}
    {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(210)|float %}
    # Start bed heating
    M140 S{BED_TEMP}
    # Use absolute coordinates
    G90
    # Reset the G-Code Z offset (adjust Z offset if needed)
    SET_GCODE_OFFSET Z=0.0
    # Home the printer
    G28
    # Move the nozzle near the bed
    G1 Z5 F3000
    # Move the nozzle very close to the bed
    G1 Z0.15 F300
    # Wait for bed to reach temperature
    M190 S{BED_TEMP}
    # Set and wait for nozzle to reach temperature
    M109 S{EXTRUDER_TEMP}

[gcode_macro END_PRINT]
gcode:
    # Turn off bed, extruder, and fan
    M140 S0
    M104 S0
    M106 S0
    # Move nozzle away from print while retracting
    G91
    G1 X-2 Y-2 E-3 F300
    # Raise nozzle by 10mm
    G1 Z10 F3000
    G90
    # Disable steppers
    M84

# See the sample-lcd.cfg file for definitions of common LCD displays.
``

SvoyakMartin, 2024-04-14:
Greetings!
I also turned out to be inattentive.
I have BTT SKR MINI E3 V1.2 and BTT
MINI 12864 V2.0.
I just can't connect them. I tried different pinouts and firmware options.
Or cyclic clicks and a blank screen without backlight, or only a working encoder and backlight.
I would be grateful for any help!

[https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/BTT%20SKR%20MINI%20E3%20V1.2PIN.pdf](BTT SKR MINI E3 V1.2 PIN)
[https://github.com/bigtreetech/MINI-12864/blob/master/mini12864_v2.0/Hardware/MINI12864%20V2.0-Pin.png](MINI12864 V2.0-Pin)
Marlin-2.1.2.2 recommended:
[image]

bowbikes, 2025-05-14:
sorry if this is necro posting but i could really use some help.
I have been trying to connect a BTT mini 12864 V2.0 to a BTT SKR e3 V3.
I have mainly followed this guide but have also looked at this (https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/650) and the various wiring diagrams associated with it.

I have been able to get the neopixels to work and can change them via mainsail but, as you can see the display just shows static.
I am fairly certain I wired things up correctly so I'm not sure why this is happening.

[image]

[image]

Here is the relevant part of my mainsail config file, any tip or advice is super appreciated.

[display]

lcd_type: uc1701

cs_pin: PB8

a0_pin: PD6

rst_pin: PB9

encoder_pins: ^PA10,^PA9

click_pin: ^!PB5

contrast: 63

spi_software_sclk_pin: PA5

spi_software_mosi_pin: PA7

spi_software_miso_pin: PA6

[neopixel SKR_screen]

pin: PA15

chain_count: 3

initial_RED: 0.4

initial_GREEN: 0.05

initial_BLUE: 0.0

color_order: RGB

bowbikes, 2025-05-15:
If anyone else sees the static, I solved this issue by swapping the CS and A0 pins

> sorry if this is necro posting but i could really use some help. I have been trying to connect a BTT mini 12864 V2.0 to a BTT SKR e3 V3. I have mainly followed this guide but have also looked at this ([#650](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/650)) and the various wiring diagrams associated with it.
> 
> I have been able to get the neopixels to work and can change them via mainsail but, as you can see the display just shows static. I am fairly certain I wired things up correctly so I'm not sure why this is happening.
> 
> [image]
> 
> [image]
> 
> Here is the relevant part of my mainsail config file, any tip or advice is super appreciated.
> 
> [display]
> 
> lcd_type: uc1701
> 
> cs_pin: PD6 #PB8
> 
> a0_pin: PB8 #PD6
> 
> rst_pin: PB9
> 
> encoder_pins: ^PA10,^PA9
> 
> click_pin: ^!PB5
> 
> contrast: 63
> 
> spi_software_sclk_pin: PA5
> 
> spi_software_mosi_pin: PA7
> 
> spi_software_miso_pin: PA6
> 
> [neopixel SKR_screen]
> 
> pin: PA15
> 
> chain_count: 3
> 
> initial_RED: 0.4
> 
> initial_GREEN: 0.05
> 
> initial_BLUE: 0.0
> 
> color_order: RGB

________________________

radek8, 2025-05-15:
@bowbikes 
I think your wiring is wrong.

You also need to edit the Marin configuration files for the correct display and enable neopixel.
Compile your own Marlin FW.

[image]