bigtreetech/BIGTREETECH-SKR-mini-E3 #735: Source code for Firmware example binaries

Marcel-Kitzbichler, 2023-04-07:
Hello, I have been trying to get my Alfawise U30 with a bltouch working but always get a bl touch error. I have tried many different marlin configs but nothing works. The only thing that kinda started to work is the example code ender 3 with bltouch for z homing for the 3.0 board which i have. The only problem with that is that the bed of the U30 is slightly smaller. Because of that i want to ask if it would be posible to share the config files for those precompiled binaries, because there it would be easy to change the bed size.

Thanks in Advance

radek8, 2023-07-07:
Look here.
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/604

brianvoo, 2023-07-14:
Took me a while to get mine working. https://github.com/brianvoo/BIGTREETECH-SKR-MINI-E3-V3-BLTouch-ZHoming give this try.