bigtreetech/BIGTREETECH-SKR-mini-E3 #742: [FR] V2 firmrware update?

cookins, 2023-05-30:
I think its time BTT update V2 firmware to current one? last one is 2.0.8.2... its getting old.
of course, there V3 already... so V2 nobody using, right?

john-clark, 2023-12-17:
I still use it. Would be nice if they included klipper along with marlin

parhedberg, 2024-01-14:
I would love for an updated firmware.. I have tried to build one on my own, but i can't seem to get the config right?

crazytimmy, 2025-03-05:
Not going to hold my breath for a reply seeing how old this request is. I've been trying to flash Klipper firmware following printer.cfg notes with no luck. Cam here hoping they had one I could try.

https://github.com/Klipper3d/klipper/blob/master/config/generic-bigtreetech-skr-mini-e3-v2.0.cfg