bigtreetech/BIGTREETECH-SKR-mini-E3 #745: SKR-mini-E3 V3 hanging after 30 min.

wcwoott, 2023-06-12:
Replaced a V2 version with V3.  All the fans are running and the house is nice and cool, so unless there is a new cooling requirement it should work.  But what I get is after approx 30 min the print just stops.  No error code on the screen, if I try to so something like stop the print then move z axis up, I get the error System is processing..  
I can do little prints but if it runs longer it hangs, I tried it with two different prints.

Any Ideas?

using: firmware-ender3-bltouch.bin    V3.0

stefi01, 2023-06-17:
I get the same happen to me, does not matter if latest Marlin as of June 2023 bugfix or latest stable, sd car on board or the touchscreen, also same with octoprint and via pc with , the board is junk

wcwoott, 2023-06-17:
Update the TFT firmware, fixed my issue.

stefi01, 2023-06-17:
Tried that but no difference, even disconnected the screen altogether and printed through Repetier but still the same result, brand new board too, shame BTT never do any testing on their products before they sell them.