bigtreetech/BIGTREETECH-SKR-mini-E3 #751: [BUG] No Printer Attached - SKR Mini E3 V2 and TFT35-E3 V3.0.1

JLiraEJ8, 2023-08-22:
Hello everyone, I'm still a newbie in this domain and I would appreciate it if you could confirm my doubt.
I have an Aquila X2 and I installed a SKR Mini E3 V2 and a TFT35-E3 V3.0.1 with BLTouch and i updated the firmware.
Everything worked perfectly until I changed the support of the TFT that I had printed and after that the message "No Printer Attached" appeared and the screen on the right side started to heat up a lot and a stain appeared. I bought another TFT and the same thing happens.
Could it be that when I installed the TFT in the support when tightening the screws it crushed some wire? I've tested with others and the same thing happens.
I think it was the motherboard that failed but it's work whell in Marlin mode (with the TFT cable disconnected because it heats up the screen a lot)
Does anyone know the voltage of the different TFT pins on the motherboard Or what could have happened apart from what I described?
Thank you for your help.

[image]
[image]
[image]
[image]

JLiraEJ8, 2023-08-22:
Problem Solved

closed 2023-08-22 by JLiraEJ8

akkoaskr, 2023-08-31:
How was the problem solved?

JLiraEJ8, 2023-09-14:
Hello, sorry for the delay. 
As you can see in one of the photos, on the TFT35 the LCD cable is not connected in the right plug (connected in the SWD) the right plug is RS232... my fault.