bigtreetech/BIGTREETECH-SKR-mini-E3 #754: SKR Mini E3 V3 does not work with CR Touch and either of the BLtouch firmware

Cr0nJ0b, 2023-10-03:
I'm looking for support or at least and acknowledgement that the firmware is borked.
### Description

### Steps to reproduce

1. Load firmware that reads [firmware-ender3-bltouch-for-z-homing.bin](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/firmware-ender3-bltouch-for-z-homing.bin):
2. Remover z-stop cable
3. init eeprom
4. rename and load firmware onto the board
5. verify that the menu for BLtouch exists -- it does
6. Home all

**Expected behavior**
All axes should home correctly to zero.  Specifically, the x and y should home, then the head should lift 11mm or so and move to the center of the bed.  The probe should extend and the z axis should lower until the probe touches the bed and activates the sensor, verifying Z home.

**Actual behavior**
X and y home, but then at whatever height the head is at it will lift and center (height will be 11mm+ original height).  the probe will extend from the sensor.  z axis will descend 11mm or so and the sensor will pull in, drop out again and pull in.  At this point the menu will read STOPPPED.

### Additional Information
I'm using the Stock firmware provided by BTT.  I did not roll this firmware.  If I had to guess there is either an issue with the board processor and wiring between the CR touch and the BL touch.

Staticshift15, 2023-10-08:
My Ender 5 pro is doing exactly the same thing

aero-gro, 2023-10-10:
Same problem here, updating an Ender 3.

kirgy, 2023-10-16:
Same issue, unable to operate ender 3 pro with upgrade to SKR mini e3 3.0.

Stoutscientist, 2023-10-24:
Same, for a $50 board they really don't know how to fix their bugs, I have seen reports of this for ~3 years

Stoutscientist, 2023-10-24:
Also, the firmware hasn't been updated in over a year...

Davmaker, 2023-11-15:
YO TAMBIEN E TENIDO PROBLEMAS CON LA MINI E3 V3-0 Y EL BLTOUCH , ADEMAS DE QUE CPMPRE UNA PANTALLA TFT35 3.0 , SUPUESTAMENTE ES COMPATIBLE CON CREALITY ENDER 5 PLUS. DECEPCION AL VER QUE EL MODO TACTIL NO CONECTA CON LA IMPRESORA , UNICAMENTE PUEDO USAR EL MODO MARLIN, SI ALGUIEN PUEDE AYUDAR LO AGRADECERÉ MUCHISIMO!!! GRACIAS

kirgy, 2023-11-15:
To solve this, I learnt how to compile the firmware by following a number of youtube videos around "marlin ender 3 firmware compilation"

Out of the back of that I created this repo which does that firmware compiling for my specific scenario. All I can suggest for others is you follow these tutorials to do it yourself - given the lack of support from bigtreetech.

https://github.com/kirgy/marlin-firmware-ender-3-pro-skr-mini-e3-crtouch

wildseyed, 2023-12-24:
> To solve this, I learnt how to compile the firmware by following a number of youtube videos around "marlin ender 3 firmware compilation"
>
> Out of the back of that I created this repo which does that [...]

Has anyone had luck recently trying to compile? Seems like Marlin AutoBuild is broken, asuming the source above did compile at some point. It's been three weeks since I put the new board in my ender. Nothing but regrets ATM.

Stoutscientist, 2023-12-29:
> Has anyone had luck recently trying to compile? Seems like Marlin AutoBuild is broken, asuming the source above did compile at some point. It's been three weeks since I put the new board in my [...]

I actually did get it to work! Here's the file:

https://download1530.mediafire.com/kp3u9ry0anngT8UOndJEzNY_7YSnGBhrM4g3OZjPONxrN9pyMKAyOmX7NFpCusCKw4yJ4k8G8CmBP93ax5MWtLgourZiMP1pcIRoqVjRyEl0IkMFk9dVpJxwq5kupgoYNziX0YZ5HOyXMSW08nNlv0rjDEKuuV0G22_VCmKMRQ/h0nfbk4k4iaiki5/firmware.bin

wildseyed, 2023-12-30:
Thanks for the file! That has me back up and running with the BL.

Stoutscientist, 2023-12-30:
> Thanks for the file! That has me back up and running with the BL.
> […](#)

Awesome! I've also been having this issue so I finally decided to put some research into it and ended up finding the fix today.

Stoutscientist, 2023-12-30:
Future updates are going to my repository: https://github.com/Stoutscientist/Updated-SKR-Mini-E3-Firmware

MpSv8, 2024-01-19:
> > > To solve this, I learnt how to compile the firmware by following a number of youtube videos around "marlin ender 3 firmware compilation"
> > > Out of the back of that I created this repo which does that firmware compiling for my specific scenario. All I can suggest for others is you follow these tutorials to do it yourself - given the lack of support from bigtreetech.
> > > https://github.com/kirgy/marlin-firmware-ender-3-pro-skr-mini-e3-crtouch
> > 
> > 
> > Has anyone had luck recently trying to compile? Seems like Marlin AutoBuild is broken, asuming the source above did compile at some point. It's been three weeks since I put the new board in my ender. Nothing but regrets ATM.
> 
> I actually did get it to work! Here's the file:
> 
> https://download1530.mediafire.com/kp3u9ry0anngT8UOndJEzNY_7YSnGBhrM4g3OZjPONxrN9pyMKAyOmX7NFpCusCKw4yJ4k8G8CmBP93ax5MWtLgourZiMP1pcIRoqVjRyEl0IkMFk9dVpJxwq5kupgoYNziX0YZ5HOyXMSW08nNlv0rjDEKuuV0G22_VCmKMRQ/h0nfbk4k4iaiki5/firmware.bin
> 
> The reason it wasn't compiling was because a library was updated since that guy posted and so when it tried to download it, it was the wrong version so I just back dated it

 

hey, @Stoutscientist  are you able to make such a file for Ender 3 (no pro version)?

Stoutscientist, 2024-01-19:
> hey, @Stoutscientist  are you able to make such a file for Ender 3 (no pro version)?

I can try but tbh I have no idea what I'm doing lol

DanKreck, 2024-01-28:
I flashed that firmware, and when Level Bed, the nozzle goes passed the right side of the bed. Is this expected?

Stoutscientist, 2024-01-29:
What's your setup?

DanKreck, 2024-01-29:
> What's your setup?

I have an Ender 3 Pro, with a SKR Mini E3 V3 board, running the firmware posted above. Fixed my Z axis issues related to the CR touch, so thats good. Just noticed that while leveling the bed the nozzle goes far right. I was wondering if it was intentional, to get the CR touch probe closer to the right side, but its not much.... maybe 1/4 inch.

Stoutscientist, 2024-02-01:
> I have an Ender 3 Pro, with a SKR Mini E3 V3 board, running the firmware posted above. Fixed my Z axis issues related to the CR touch, so thats good. Just noticed that while leveling the bed the [...]

If you attach a photo I can double check but mine does something similar and prints fine so I think it's normal

Stoutscientist, 2024-02-01:
> hey, @Stoutscientist are you able to make such a file for Ender 3 (no pro version)?

I actually just read that the Ender 3 Pro and non pro use the same firmware so it should work for you

CelticNickC, 2024-02-08:
I performed the flash on ender 3, but it runs the nozzle into the plate with this firmware

Chilson3D, 2024-02-08:
mine works, ill share a snip of my .cfg I do remember reading a post that the "pin out" needs to be changed as follows:
Updated BL Touch PIN for CR touch #718
			@@ -10,7 +10,7 @@
Line 10 		# See docs/Config_Reference.md for a description of parameters.
Line 11
Line 12		[bltouch]
Line 13		sensor_pin: PC14      #(wrong code)
Line 13		sensor_pin: ^PC14    #(correct code)
Line 14		control_pin: PA1
Line 15		x_offset: -40
Line 16		y_offset: -10

**My setting:**
[bltouch]
sensor_pin: ^PC14
control_pin: PA1
x_offset: -45.5
y_offset: -9
#z_offset: 3.0
#speed:

Chilson3D, 2024-02-08:
mine works, ill share a snip of my .cfg I do remember reading a post that the "pin out" needs to be changed as follows:
**Updated BL Touch PIN for CR touch #718**
			@@ -10,7 +10,7 @@
Line 10 		# See docs/Config_Reference.md for a description of parameters.
Line 11
Line 12		[bltouch]
Line 13		sensor_pin: PC14      #(wrong code)
Line 13		sensor_pin: ^PC14    #(correct code)
Line 14		control_pin: PA1
Line 15		x_offset: -40
Line 16		y_offset: -10

**My setting:**
[bltouch]
sensor_pin: ^PC14
control_pin: PA1
x_offset: -45.5
y_offset: -9
#z_offset: 3.0
#speed:

CelticNickC, 2024-02-08:
> mine works, ill share a snip of my .cfg I do remember reading a post that the "pin out" needs to be changed as follows:
>
> **Updated BL Touch PIN for CR touch #718**
>
> 			@@ -10,7 +10,7 @@
>
> Line 10 		# [...]

So I went through and attempted to edit coding while following videos. Nearly all of the codes changed were in far different spots due to the updating on the file being used compared to the video. I’m still coming back with the same issue even after seeing changes validated from the edit. I had to pull the bin file from vsc and drag it onto my sd card. I’m using an old MacBook to try to program these codes, and the file just wasn’t showing up anywhere even if I had zero errors. 

The error showing up after the bltouch touches down says “error: 
!! STOP called because of BLTouch error -restart with m999”

Any actions to take for this or ideas of code to tweak? I was not able to locate in the coding where your recommendations for pins would be

topybear, 2024-02-08:
> Any actions to take for this or ideas of code to tweak? I was not able to locate in the coding where your recommendations for pins would be

Not in code. Printer.cfg file.

CelticNickC, 2024-02-08:
> > Any actions to take for this or ideas of code to tweak? I was not able to locate in the coding where your recommendations for pins would be
> 
> 
> 
> Not in code. Printer.cfg file.

Would I need to add this file to the build to be able to adjust? Or where would I find the printer.cfg file?

config/Ender-3 Pro/SKR-Mini-E3-V3.0/printer.cfg

topybear, 2024-02-08:
This is Klipper config file.

CelticNickC, 2024-02-09:
> This is Klipper config file. 

Would it work the same as for marlin?

DINOS-DEAD, 2024-02-15:
I got fet up and just went to TH3Dstudio payed the five bucks and finished the config.h file as instructed and add my features, compiled and now up and running less than an 30 minutes

Seelenkind, 2024-07-24:
try this, my new updated firmware with actual Marlin 
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/pull/776

raff17, 2024-11-29:
> try this, my new updated firmware with actual Marlin #776

I recently tried it and my cr touch will move like its working but when it touches the printing bed it doesn't retract causing the cr touch to fight the z motor causing the printer to vibrate. I thought that possibly it could be a z-offset thing, but even when adjusting the result is the same.

hikita3318, 2025-03-19:
I got a second hand ender3 with a SKR mini E3 V3 preinstalled few weeks back.
When I updated to the latest firmware, I had the touch sensor was not working
properly and the nozzle grinding the build plate... :-(

Seems that the previous owner was using a **clone BL/CR sensor**.
" disabling BLTOUCH_FORCE_SW_MODE In Configuration_adv.h" 
and rebuilding the firmware worked for me. YMMV.

https://github.com/MarlinFirmware/Marlin/issues/25588#issuecomment-1493369831