bigtreetech/BIGTREETECH-SKR-mini-E3 #755: [FR] Add the source code used for building firmware

medisoft, 2023-10-27:
**Is your feature request related to a problem? Please describe.**
I'm trying to add support for the filament sensor to the board, because the available firmwares only include bltouch, but the filament sensor is not there. I downloaded latest version of Marlin, 2.0.x, and 2.1.x, normal and bugfixes, and used the configuration for Ender 3 Pro with BTT SKT Mini E3 V3.0. Everything builds OK, but when I try to flash the firmware, the printer stays with blank screen and the firmware.bin is not moved to firmware.cur, after that, the printer don't boot anymore, unless I flash the https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin/firmware-ender3.bin or any of the other firmwares in that repo.

**Describe the solution you'd like**
So I think you should make the source you used available in the same folder as the binary files, so that if one needs to add or remove something, it is possible.

**Describe alternatives you've considered**
I built the firmware using the configurations at https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3%20Pro/BigTreeTech%20SKR%20Mini%20E3%203.0, for release, imported and bugfix, for 2.0.x and 2.1.x, and always the same result

yasavvym8, 2023-11-23:
I'd also appreciate it if they actually kept their repository up to date

john-clark, 2023-12-17:
No current support = no future purchases.