bigtreetech/BIGTREETECH-SKR-mini-E3 #757: blue screen

TsReflexx, 2024-01-17:
tried downloading new firmware to my ender 3 and now I’m just getting a blue screen and it’s not coming on can anyone help me please
i have the skr mini v2 and bl touch 
i had it working then the level wasnt working properly so i deleted everything off my drive and tried to redownload it and now i just get a blue screen and its not working at all

doug1261, 2024-02-02:
Similar issue. Ender 3 with skr mini v2 and cr touch. Uploaded firmware.bin and now BLUE screen.

tomphale, 2024-02-04:
I have exactly the same issue Ender 3 Pro, Skr mini e3 v3. Updated Firmware now just a blank screen, Does anybody have a solution?

doug1261, 2024-02-05:
Fortunately I can return mine to Amazon

otacon6530, 2025-02-21:
I have been having the same issue. For some reason plugging in a computer via the usb (octoprint) brought the screen back, but I am still seeing the original firmware listed on the device "bugfix-2.0.X".