bigtreetech/BIGTREETECH-SKR-mini-E3 #769: Provide configuration files used to build firmware V3.0 Z-homing

JuliBos, 2024-05-24:
I cannot customize the firmware for my printer, because the "Configuration.h" and "Configuration_adv.h" files are not provided in the repository.

Describe the solution you'd like
Please provide the "Configuration.h" and "Configuration_adv.h" files used to compile the "firmware-ender3-bltouch-for-z-homing.bin" in the "BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin)" directory.

Seelenkind, 2024-07-21:
Hi , the last merged update was from mine configurations . Look at here 
https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin/Ender%203

Seelenkind, 2024-07-24:
Look here for new Marlin Version 
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/777