bigtreetech/BIGTREETECH-SKR-mini-E3 #771: [FR] TTL control of the laser module with SKR e3 mini v2

YaBrick, 2024-06-20:
Hello. I am facing a problem - I have a laser (LS-40W-F23) which is controlled by TTL signal. 
One of the options I have seen is to solder to the FAN1 pin going to the fan control mosfet, and by setting the fan load percentage in gcode ( control the laser. 
However, I'm not much of a solder guy, and I'm sure many people would like to be able to connect the laser quickly and easily.
Is it possible to use, for example, pin PA8 from the Neopixel connector? 

#define SPINDLE_LASER_PWM_PIN 8 // Example for defining the pin in case it could work. By the Marlin's documentation, should be placed in Configuration_adv.h

Will this definition allow to control the laser directly connected by to this pin? 

Sincerely yours
Eric

YaBrick, 2024-06-21:
In case someone would like to use Neopixel's PWM as control pin for PWM of the laser (LS-40W-F23):
For now I've made these changes to the Marlin firmware:
1. Uncomment #define DIRECT_PIN_CONTROL in Configuration_adv.h
2. Uncomment #define PINS_DEBUGGING to use M43 for pins debugging purposes
3. Connect PWM of the laser to the PA8 pin on the board WITH CURRENT LIMITING RESISTOR 150ohm

Now, giving the M42 P8 S255 (P8 bcause that's the correct command) (S - PWM value you'd want to give to the laser) gives full PWM signal. Values between 0 and 255 are supported. In case you'd want this solution - replace M3 (usual Marlin laser controlling comand) with M42 command in the postprocessor, or well... Manually replace every M3 by M42 P8. They should have the same agruments, 
for example, M3 S128 = M42 P8 S128

HAVENT TESTED IT ON A LASER. Yet. I've only verified if there's correct output on PA8 pin, and there is, but there also a question of the laser's PWM frequency, sometimes it need frequency much bigger than board can provide.

YaBrick, 2024-06-21:
So, if you want to CORRECTLY use the M3 command to control PWM of the laser you should know - by default after setting the M3 command it won't work. I'm not sure why, but when you use "I" argument (M3 I S255) it gives PWM on PA8 AFTER getting the G1/G2/G3 command (any moving command I guess). It is called "Inline Mode", you can check the info here: https://marlinfw.org/docs/configuration/2.0.9/laser_spindle.html
Basically is is as safe measure to start the laser only when it starts moving. But you can ucomment the     
#define LASER_POWER_SYNC in Configuration_adv.h to disable it, so without "I" argument it flashes immideatly

YaBrick, 2024-06-21:
So, I've tested the setup of the laser with SKR e3 mini v2, and it's working. 
What you should you do to get my result (I'm supposing you do already have working Marlin's distributive for this motherboard):
1. In configuration_adv.h uncomment #define LASER_FEATURE
2. In the line ABOVE add #define SPINDLE_LASER_PWM_PIN  PA8 
3. Uncomment  #define LASER_POWER_SYNC
4. Compile and upload the firmware.bin to the motherboard
5. Connect laser's PWM with PA8 pin VIA 150 ohM CURRENT LIMITING RESISTOR, without it I am afraid you can burn the Neopixel's buffer chip (PA8 pin is marked on the back of the board, or you can find it by the pinout of the board - https://docs.vorondesign.com/build/electrical/v0_miniE3_v20_wiring.html)
6. Connect the GND of the Laser with the GND of the GND on the Neopixel connector
7. Optionally, uncomment #define DIRECT_PIN_CONTROL so you can enable laser with M42 P8 S[0-255] gcode command. 

Why?
Because when you use M3 command (see M3 command here: https://marlinfw.org/docs/configuration/2.0.9/laser_spindle.html)
laser will work for 1000ms (can change it in LASER_SAFETY_TIMEOUT_MS) is there's no G1/G2/G3 command (G0 doesn't count!!!)
So if you want to test it right then - you'd better use M42 P8 S[0-255] command to directly put the PWM signal without any safety measures. I'm considering you're careful enough to test it with all the caution.

Example use of the laser in engraving:
G1 X100 S[0-255] will make a straight line (if X == 0 and the start)
M3 S[0-255] will make a dot by striking the laser at one point for 1000ms and then turn off
M42 P8 S[0-255] will enable at the with S PWM signal. Won't turn off by itself

Marc--, 2024-07-26:
@YaBrick Hi there, sorry to contact you this way, but i found your repo about this project and have the exact same !
I have some problems and would like to exchange on it, but can't open issues on your repo. Is it possible to chat ?

YaBrick, 2024-07-26:
Hello Marc! I'll be happy to help you, didn't know that issues were closed on this repo, I'll open then now