bigtreetech/BIGTREETECH-SKR-mini-E3 #775: No firmware for STM32G0B0RET6 ?

sudo-su-root, 2024-07-24:
Bought BIGTREETECH SKR Mini E3 V3.0 from BIGTREETECH store on Amazon UK

Connected board and Marlin was showing fine and I could probe the MCU from the pi via USB. Downloaded the correct prebuilt firmware from [github](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Klipper/firmware-USB.bin) and I could tell it flashed to some degree as screen was now all blue no longer showing Marlin. From this point onwards I cannot probe MCU via USB.

The description on amazon states STM32G0B1RET6 but on inspection it's been shipped with a STM32G0B0RET6.

Flashed the Marlin firmware from [github](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/firmware-ender3.bin) even though it's probably for the wrong MCU and the screen comes back to life with Marlin but still cannot probe MCU via USB.

I have watched a good 10 videos covering the creation of firmware, followed the standard settings for STM32G0B1RET6 but changed just the MCU to STM32G0B0RET6 and it still does not fix the issues.

I suspect at this point the usb port is faulty but it would be nice if someone were to compile and upload a tested firmware.bin for this MCU. Maybe there is no real difference and the compiled firmware should be working fine. despite wrong MCU selection.

sudo dmesg -w output.....

[   48.905983] Indeed it is in host mode hprt0 = 00021501
[   49.093787] usb 1-1: new full-speed USB device number 2 using dwc_otg
[   49.094255] Indeed it is in host mode hprt0 = 00021501
[   49.281841] usb 1-1: device descriptor read/64, error -71
[   49.390237] Indeed it is in host mode hprt0 = 00021501
[   49.577785] usb 1-1: device descriptor read/64, error -71
[   49.685902] Indeed it is in host mode hprt0 = 00021501
[   49.873868] usb 1-1: new full-speed USB device number 3 using dwc_otg
[   49.874485] Indeed it is in host mode hprt0 = 00021501
[   50.061878] usb 1-1: device descriptor read/64, error -71
[   50.170388] Indeed it is in host mode hprt0 = 00021501
[   50.361847] usb 1-1: device descriptor read/64, error -71
[   50.469964] usb usb1-port1: attempt power cycle

dkarpes, 2024-08-13:
Hi, you can check mines: https://github.com/dkarpes/Marlin-2.1.x-BTT-SKR-MINI-E3-V3-TFT35-BLTOUCH

lwq690, 2024-10-08:
hello  have your question have been solved? i have a photo , 
[image]

lwq690, 2024-10-08:
[image]

puzari-amit, 2024-10-09:
Hello, thank you for the response.

With regard to the picture you have shared, where is this picture from?

Where does this code need to be implemented?

puzari-amit, 2024-10-10:
Hello,

I think the image you shared is klipper.

I am looking for marlin source code.

changwang31, 2025-04-10:
i am also having this issur if anyone could help