bigtreetech/BIGTREETECH-SKR-mini-E3 #777: [FR] Marlin 2.1.2.4 for SKR Mini E3 V3.0 STMG0B1RE with BLTOUCH for Zhoming

Seelenkind, 2024-07-24:
Hi guys. I have uploaded the latest Marlin 2.1.2.4 for SKR Mini E3 V3.0 with STM32G0B1RE with BLTOUCH to Github.
If anyone wants to try it out, please give feedback.

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/0c7333dc19e4b7744f7e288de309cd8e83550580/firmware/V3.0/Marlin

averagecanadian, 2024-08-07:
I tried this firmware. My Z axis won't go down, only up. Z axis still works with default firmware. Any ideas?

Seelenkind, 2024-08-07:
Pls unplug Z axis on Board . BL touch is new Z axis endstop Switch

averagecanadian, 2024-08-08:
Thanks, rookie mistake. I had the BLTouch in the z stop since the 5 pin didn't work with the original firmware. Thank you.

dbyy, 2024-11-26:
> Hi guys. I have uploaded the latest Marlin 2.1.2.4 for SKR Mini E3 V3.0 with STM32G0B1RE with BLTOUCH to Github. If anyone wants to try it out, please give feedback.
> 
> https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/0c7333dc19e4b7744f7e288de309cd8e83550580/firmware/V3.0/Marlin

My board has a STM32G0B0RE CPU. I'm assuming I can test the new version as well with this CPU.

Leo88-max, 2025-01-10:
Hello, since I installed Marlin version 2.1.2.4 for SKR Mini E3 V3.0 on my Ender3 at the time of printing the Z while working stops without error message??? if I turn the machine off and turn it back on it resumes printing and several shutdowns are random. I replaced the 24V power supply and the motherboard and even with the new card I have the problem. I tested several firmware and it does not solve the problem. Can you help me to solve what is wrong. THANKS.

jcase16, 2025-03-11:
I haven't flashed my ender 3 with this yet, but I'm wondering if it handles a larger mesh size?  I've been looking for a firmware that does BLTouch and UBL so I can have many more points in my mesh since my bed is 400X400. Any ideas?