bigtreetech/BIGTREETECH-SKR-mini-E3 #779: [FR] Ender 6 firmware for Marlin

mreverett, 2024-08-04:
I've had a really hard time getting firmware for the BTT skr mini e3 v3.0.1 for the Ender 6.

This repo only goes up to the Ender 5. 

It would make my day if there was some available firmware for the Ender 6 on here!

xXMephisto82Xx, 2024-10-02:
Hy frage hast du vielleicht eine funktionierente firmware für BTT skr mini e3 v3.0 ender 6 lg

mreverett, 2024-10-02:
entschuldigung, ich habe die firmware nicht mehr, sie hat nicht richtig
funktioniert

xXMephisto82Xx, 2024-10-02:
ok danke

xIExodusIx, 2024-12-03:
@xXMephisto82Xx 
I have a working Marlin 2.1.2.5 firmware for the Creality Ender-6 with BTT SKR E3 Mini V3.0 motherboard & BTT TFT35 V3.0.1 display.
A working Marlin 2.1.3-Beta1 firmware for the Ender-6 with BTT SKR E3 Mini V3.0 & BTT TFT35 V3.0.1 is also already available.
If you are still interested, just get in touch.